autowarefoundation/autoware-documentation #5: Revisit documentation tree structure for more usability

kenji-miyake, 2021-12-21:


kenji-miyake, 2021-12-21:
I'd like to re-organize this into like this.

- Introduction
  - About Autoware
  - About this documentation
- Installation
  - Supported platforms
  - Installation of Autoware
    - Local
    - Docker
  - Installation of tools for users
  - Installation of tools for developers
- Tutorials
  - Planning simulation
  - Log data simulation
  - E2E simulation
  - Scenario testing
- How-to guides
  - Integration
    - Hardware setup
      - Vehicle setup
      - ECU setup
      - Sensors setup
    - Software setup
      - Customizing
      - Tuning
      - Map creation
  - Write a node
  - How to debug
  - etc.
- Design
  - Overview
  - Component interfaces
  - AD API
  - Repository structure
  - Development process
  - Release process
- Contributing
  - Overview
    - License / code of conduct
    - Contribution workflow
  - Coding guidelines
    - Specific languages
      - C++
      - Python
      - etc.
    - ROS node design
      - topic namespace
      - launch
      - parameter
      - console logging
      - etc.
  - Testing guidelines
  - Documentation guidelines
  - Pull request guidelines
- Help
  - Support guidelines
  - Links to related documents

kenji-miyake, 2021-12-21:
We'll create an empty tree like the above and consider each content in #6.

kenji-miyake, 2021-12-21:
Related: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/1178

mitsudome-r, 2021-12-22:
From: https://gitlab.com/autowarefoundation/autoware-foundation/-/issues/192
  * For Tuning section, we should think of a way of describing tuning instruction in design document of each package and the Tuning guide in autoware-documention should be a collection of links to the package instruction. Same for customization. 
  * We don't want to have these documents heavily dependent on implementation since it is difficult to keep documents up-to-date while we do the development.

PR #14 (2022-01-21): docs: add documentation tree structure
Resolves #5
closed 2022-01-21 by kenji-miyake