autowarefoundation/autoware-documentation #6: Revisit development rules for more efficiency

kenji-miyake, 2021-12-21:


isamu-takagi, 2021-12-23:
I think we need rules to clarify the interface for the following reasons.
- It's often difficult to know what parameters/topic the node has because the interface definitions (e.g. create_subscription) are scattered in the code.
- There is no way to know which topics are public or private in the launch file.
- A mixture of the default values in the source code and the remap values in the launch file.

IshitaTakeshi, 2022-02-02:
Not a strict rule, but there's a guideline to write good code, called [Object Calisthenics](https://www.bennadel.com/resources/uploads/2012/objectcalisthenics.pdf). This is a pretty strict guideline, so it is not necessary to precisely follow, but it helps to write testable, reusable, and extensible code.

IshitaTakeshi, 2022-02-02:
Anyway, whatever we adopt, it would be better to have some tips or guidelines for writing good code other than coding rules.
By accumulating knowledge and tips and publishing them, we can improve our product quality and contribute to open source communities.

kenji-miyake, 2022-02-02:
> Not a strict rule, but there's a guideline to write good code, called Object Calisthenics. This is a pretty strict guideline, so it is not necessary to precisely follow, but it helps to write testable, reusable, and extensible code.

In that case, it's better to link the document as a reference.
Regarding some important techniques can be picked up and be written in our guidelines. (If there are no license issues.)

> Anyway, whatever we adopt, it would be better to have some tips or guidelines for writing good code other than coding rules.

Yes, of course. I'm preparing it.

kenji-miyake, 2022-05-05:
I think I can write some more documents this month.