autowarefoundation/autoware-documentation #84: Add a documentation guideline

IshitaTakeshi, 2022-04-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

I want to add a documentation guideline in the following content to make the instructions clear.

### Purpose

Provide clear contribution instructions to the developers

### Possible approaches

Add the following content to the documentation guideline page

# Documentation guidelines

## Instructions to add documentation

* If you have knowledge about Autoware usage or functionality, or if you have a question about Autoware, and once it is resolved, turn the knowledge into the documentation!

There are two ways to contribute to the documentation page.
1. If you can make a documentation page based on your knowledge, please [send a pull request](https://github.com/autowarefoundation/autoware-documentation/pulls). If you are unaware of if you can open a pull request or unaware of the format, you can ask on the [feature requests page](https://github.com/autowarefoundation/autoware/discussions/categories/feature-requests).
2. If you are curious about what to write or whether your contribution is worthwhile or not, you can ask on [Q&A page](https://github.com/autowarefoundation/autoware/discussions/categories/q-a).

### Definition of done

* Developers can clearly understand the contribution instructions
* Cover basic cases, such as in which case to open a pull request, or in which case to open a discussion

kenji-miyake, 2022-04-18:
@IshitaTakeshi Thank you for opening an issue. I have some comments.

> If you have knowledge about Autoware usage or functionality, or if you have a question about Autoware, and once it is resolved, turn the knowledge into the documentation!

I think this part should be in https://autowarefoundation.github.io/autoware-documentation/main/help/support-guidelines/ or https://autowarefoundation.github.io/autoware-documentation/main/contributing/.

> If you can make a documentation page based on your knowledge, please [send a pull request](https://github.com/autowarefoundation/autoware-documentation/pulls). If you are unaware of if you can open [...]

The feature request category isn't for documentation.

kenji-miyake, 2022-04-18:
Also, could you list up what is unclear with the current documentation?
For example,
- Users can't understand how to submit a PR about `...`.

IshitaTakeshi, 2022-04-18:
> If you have knowledge about Autoware usage or functionality, or if you have a question about Autoware, and once it is resolved, turn the knowledge into the documentation!

In which part of the contribution page should I place it?

IshitaTakeshi, 2022-04-18:
> Also, could you list up what is unclear with the current documentation?

I was unsure about how to make a new page in the documentation. I wanted to add a page that explains how to initialize the NDT pose. But it was unclear how to add a documentation page, so I want to update the contribution guideline.

kenji-miyake, 2022-04-18:
> 
If you have knowledge about Autoware usage or functionality, or if you have a question about Autoware, and once it is resolved, turn the knowledge into the documentation!

In which part of the contribution page should I place it?

I personally feel:

- Regarding `If you have knowledge about Autoware usage or functionality`, https://autowarefoundation.github.io/autoware-documentation/main/contributing/#contribution-workflows

- Regarding `if you have a question about Autoware, and once it is resolved, turn the knowledge into the documentation!`, https://autowarefoundation.github.io/autoware-documentation/main/help/support-guidelines/#github-discussions

> I was unsure about how to make a new page in the documentation. I wanted to add a page that explains how to initialize the NDT pose. But it was unclear how to add a documentation page, so I want to update the contribution guideline.

In that case, I feel it's covered by `If you are curious about what to write or whether your contribution is worthwhile or not, you can ask on [Q&A page](https://github.com/autowarefoundation/autoware/discussions/categories/q-a).`, right?

I feel it should be placed in https://autowarefoundation.github.io/autoware-documentation/main/contributing/#contribution-workflows.

IshitaTakeshi, 2022-04-18:
> In that case, I feel it's covered by If you are curious about what to write or whether your contribution is worthwhile or not, you can ask on [Q&A page](https://github.com/autowarefoundation/autoware/discussions/categories/q-a)., right?

That's why I opened this pull request first.

IshitaTakeshi, 2022-05-06:
I close this issue as the documentation has been largely updated in https://github.com/autowarefoundation/autoware-documentation/pull/91

closed 2022-05-06 by IshitaTakeshi