autowarefoundation/autoware-documentation #102: Update Open AD Kit wiki

kasperornmeck, 2022-05-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Update the Open AD Kit wiki with more details of e.g., the benefits of developing and deploying Autoware as a microservices architecture

### Purpose

Provide more clarity and raise awareness of the benefits microsevices architecture bring to Autoware 

### Possible approaches

Update the wiki https://github.com/autowarefoundation/autoware-projects/wiki/Open-AD-Kit-working-group

### Definition of done

Once changes have been agreed on, reviewed and merged.

kasperornmeck, 2022-05-25:
# Open AD Kit Working Group

Inspired by [Scalable Open Architecture For Embedded Edge’s (SOAFEE)](https://soafee.io/) take on how to enable the Software Defined Vehicle (SDV), Open Autonomous Drive Kit (Open AD Kit) aims to democratize autonomous drive (AD) systems by bringing the cloud and edge closer together. In doing so, lowering the threshold of developing and deploying the Autoware software stack.

## Working group leaders

- Armağan Arslan (LeoDrive.ai)
- David Walmroth (Ann Arbor Autonomous Vehicle Group)
- Lalith Vipulananthan (Tier IV)

<details>
<summary>Former leaders</summary>

- Matt Spencer (Arm)
</details>

## Working group goals

The Open AD Kit working group's main goals are
- introduce modern cloud-native development methodologies in Autoware's CI/CD
- add virtualization in different levels in the software stack
    - hypervisor to run VMs
    - containers to run applications
- deploy Autoware as a microservices architecture
    - deployment both in the could and at the edge
    - an orchestrator will manage containers
    - the orchestrator will be mixed critical aware
- providee a consistent production environment in which to deploy Autoware by leveraging [Edge Workload Abstraction and Orchestration Layer (EWAOL)](https://ewaol.sites.arm.com/meta-ewaol/overview.html)
    - EWAOL is a hardware-agnostic custom Linux distro built using Yocto

Benefits of Open AD Kit:
- a developer will be able to add their software to or alongside Autoware
- Autoware will easily 
- most testing will be done in the cloud, reducing the need for testing at the edge

### Open AD Kit Architecture
[image]

### Out of Scope
The Open AD Kit working group does not develop software; it takes existing software in Autoware Core/Universe and deploys it, 

## Current activities

Overview:
- Planning the scope of Open AD Kit 2.0
- Adding new features to the CI/CD enabling multi-container build pipeline
- ROS packages dependencies

See [Open AD Kit project page](https://github.com/orgs/autowarefoundation/projects/6) for on-going tasks.

## Meeting minutes
From March 10th 2022 onward, meeting minutes can be found in GitHub discussions with the [Meeting:openadkit_wg](https://github.com/autowarefoundation/autoware/discussions/categories/workgroup-meetings?discussions_q=category%3A%22Workgroup+Meetings%22+label%3A%22Meeting%3A+openadkit_wg%22) label.

<details>
<summary>Archived Meeting minutes from GitLab</summary>

- [2022/02/24](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20220224)
- [2022/02/10](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20220210)
- [2022/01/26](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20220126)
- [2022/01/12](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20220112)
- [2021/12/16](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20211216)
- [2021/12/01](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20211201)
- [2021/11/03](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20211103)
- [2021/10/20](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20211020)
- [2021/10/06](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20211006)
- [2021/09/22](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20210922)
- [2021/09/08](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20210908)
- [2021/08/24](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20210824)
- [2021/07/28](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20210728)
- [2021/07/14](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20210714)
- [2021/07/01](https://gitlab.com/autowarefoundation/autoware-foundation/-/wikis/open-ad-kit/minutes/20210701)
</details>

## How to participate

The Open AD Kit working group meets every week on a rotating schedule as follows:
- Slot 1: Wednesday 10pm UTC (chair: David)
- Slot 2: Wednesday 3pm UTC  (chair: Armağan)

For meeting information, see the [Autoware Foundation event calendar](https://calendar.google.com/calendar/embed?src=autoware.org_6lol0ho5ft0217h8c60pi1fm30%40group.calendar.google.com&ctz=Asia%2FTokyo) for when upcoming meetings will be held. 

If you use Google Calendar, you can subscribe to this calendar directly to have it displayed alongside your own calendar events. To do this, click the "+ Google Calendar" link in the bottom right corner of that site. Note that this will *not* add the events to your own calendar, meaning you may not receive notifications about meetings starting soon. For those not using Google Calendar, you can add the calendar to your own by downloading it as an ICS file and importing that into your calendar software.

LalithVipulananthan, 2022-05-31:
@samet-kutuk Just to confirm, were you able to work with @kasperornmeck on the proposed changes? Failing that, have you had time to review the proposed changes to give any feedback?

kasperornmeck, 2022-06-02:
@LalithVipulananthan 
I updated the description, making some clarifications I realized were needed after the WG call this week. I'm happy with the state of it now.

samet-kutuk, 2022-06-06:
> @samet-kutuk Just to confirm, were you able to work with @kasperornmeck on the proposed changes? Failing that, have you had time to review the proposed changes to give any feedback?

Hi Lalith, I just had the time to look into it.

When I previously looked at it there were some unfinished sentences but Kasper fixed that with revision. I have no feedback to provide as I believe it's sufficient for Github purposes.

kasperornmeck, 2022-06-06:
@samet-kutuk 
Thanks for reviewing the updated revision, and I did indeed update it after our call last week. Girish has reviewed this and had no comments.

@LalithVipulananthan 
If there are no other reviews pending, I think we can update the Open AD Kit page?

NaoyoriTanzawa, 2022-06-07:
@kasperornmeck Thank you for the proposal!
Can we leave "Relationship between Autoware, Reference Design, and Open AD Kit" and "Proposal of Detailing The Autoware High Level ODD Roadmap" from the current one?

LalithVipulananthan, 2022-06-07:
@kasperornmeck The plan was to wait until this week's WG meeting for any other feedback, so if we have received no other comments by that time, then the wiki can be updated. 

And just in case, please note Tanzawa-san's comment above about leaving in the slides from the current wiki page (@samet-kutuk).

LalithVipulananthan, 2022-06-13:
Updated the wiki with the agreed draft above, so closing out issue

closed 2022-06-13 by LalithVipulananthan