autowarefoundation/autoware-documentation #114: Document the usage of calibration tools (with a example rosbag if possible)

mitsudome-r, 2022-05-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description
Sub ticket of https://github.com/autowarefoundation/autoware.universe/issues/592
Add documentation for using calibration tools

### Purpose

Create instructions for calibration tool so that users can generate calibration parameters for sensors.
This includes: 
 - Intrinsic parameter for cameras
 - extrinsic parameter between cameras (camera-camera calibration)
 - extrinsic parameter between lidars (lidar-lidar calibration)
 - extrinsic parameter between a camera and a lidar (camera-lidar calibration)

### Definition of done

- [ ] Documentation of intrinsic camera parameters available
- [ ] Documentation of camera extrinsic parameter calibration (camera-camera calibration) available
- [ ] Documentation of extrinsic parameter between lidars (lidar-lidar calibration) available
- [ ] Documentation of extrinsic parameter between a camera and a lidar (camera-lidar calibration) available

Sharrrrk, 2022-05-23:
Infomation: 
Modified branch of an open source LiDAR-IMU calibration tool: https://gitlab.com/JianKang_Egon/li_calib/-/tree/main/
Documentation is updated to the README.

JianKangEgon, 2022-06-09:
To calibrate intrinsic parameter of camera, ROS1 provides one package to make it:
http://wiki.ros.org/camera_calibration
It's been updated to ROS1 Noetic, currently not for ROS2

mitsudome-r, 2022-07-26:
We can add documentation for https://github.com/autowarefoundation/autoware/discussions/2727