autowarefoundation/autoware-documentation #123: Adding more detail to the existing hardware requirements

Sharrrrk, 2022-05-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

According to the current documentation page of Autoware, little information is provided regarding the hardware requirement to run Autoware either in simulation or on real vehicle.

The only mentioned information is the target hardware and prerequisites:

Architecture
- amd64
- arm64

Prerequisites:
- Ubuntu 20.04
- git

Although Autoware is scalable and does not have a universal hardware requirment, it would be helpful to provide some reference hardware platform requirements for Autoware users.

### Purpose

Provide reference hardware platform requirements in [target platforms](https://autowarefoundation.github.io/autoware-documentation/main/installation/#target-platforms) and [vehicle integration](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/how-to-integrate-autoware-with-your-vehicle/#1-prepare-your-real-vehicle-hardware)

### Possible approaches

Describe at least one reference hardware platform for each architecture:
- amd64
- arm64

amd64 may follow the requirement form [AutowareArchitectureProposal](https://github.com/tier4/AutowareArchitectureProposal.proj/blob/main/docs/tutorial/Requirements.md)

arm64 will be a distributed hardware platform where with partial Autoware stack running on each

### Definition of done

- [ ] Reference hardware requirement for amd64
- [ ] Reference hardware requirement for arm64

LalithVipulananthan, 2022-05-24:
Based on @kenji-miyake's comment in the original hardware requirements discussion, the following requirements should be suitable to run the full Autoware stack on a target amd64 platform.

- Hardware
  - x86 CPU (8 cores)
  - 16GB RAM
  - [Optional] NVIDIA GPU (4GB RAM)
  - Performance will be improved with more cores, RAM and a higher-spec graphics card.
  - Although not required to run basic functionality, a GPU is mandatory in order to run the following components:
    - Perception/ObjectRecognition
    - Perception/TrafficLightRecognitionRecognition
- Software
  - Ubuntu 22.04
  - NVIDIA driver

armuren, 2022-05-24:
@LalithVipulananthan Ubuntu 20.04, or 22.04 is required?  Also can you please link the AutowareArchitectureProposal from @kenji-miyake

LalithVipulananthan, 2022-05-24:
> @LalithVipulananthan Ubuntu 20.04, or 22.04 is required?

For now Ubuntu 20.04 is fine, but once ROS 2 Galactic reaches end of support this November we will move to using ROS 2 Humble for which Ubuntu 22.04 is required. I guess we were getting a little ahead of ourselves. 😅

armuren, 2022-05-24:
@LalithVipulananthan many thanks for the detailed answer.  I wonder if any discussions have been made about different requirements for different ODD's.  More specifically, which Autoware Universe packages are necessary for a particular ODD vs another.   I suspect the cargo delivery ODD requires less from the perception stack than the shuttle ODD, and similarly the racing ODD.

Sharrrrk, 2022-05-25:
> Based on @kenji-miyake's comment in the original hardware requirements discussion, the AutowareArchitectureProposal requirements for amd64 is suitable to run the full Autoware stack, but should be updated to state that Ubuntu 22.04 is required.

Agree, will update to have both 20.04 and 22.04 when everything is ready.

closed 2022-06-10 by Sharrrrk