autowarefoundation/autoware-documentation #166: Update Source Installation For Jetson Users

NilaySener, 2022-07-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Update the current source installation documentation to include how to install and compile differently Autoware with arm64 hardware.

### Purpose

For arm64 users, specifying differences in installation of dependencies and changes in build phase.

### Possible approaches

[Source installation section](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/) of the document will be updated for Jetpack users.

### Definition of done

Autoware was compiled with a few different steps for Jetson that were not specified in the documentation.

idorobotics, 2023-11-04:
Linked PR is closed, other PR is merged. @xmfcx can this issue be closed?

xmfcx, 2024-03-07:
This issue is fixed by:
- https://github.com/autowarefoundation/autoware.universe/pull/1472