autowarefoundation/autoware-documentation #209: new feat: Installation of related tools

JianKangEgon, 2022-10-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

We AutoCore members are willing to contribute ourselves to preparing this part regarding: https://autowarefoundation.github.io/autoware-documentation/main/installation/related-tools/

However, we are not that clear about the definition of "related tools". Do these tools refer to some third-party tools / dependencies during installation such like pcl-tools or something else?

### Purpose

Get clear about the scope of "related tools"

### Possible approaches

TBD.

### Definition of done

Prepare docs

kenji-miyake, 2022-10-10:
@JianKangEgon Thank you for asking this.

Originally, I was thinking of E2E simulators.
https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html

But it can contain any other tools like what you mentioned.
And it's okay to rephrase the word `Related tools` to be more descriptive. (I didn't have a good idea.)

JianKangEgon, 2022-10-10:
@kenji-miyake oh thanks for the quick reply. I'll check the above link and close this issue.

closed 2022-10-10 by JianKangEgon