autowarefoundation/autoware-documentation #212: Add guides about ros2 console-logging

alanmengg, 2022-10-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Writing guidelines about ros2 console-logging.

### Purpose

Current console-logging.md in autowarefoundation/autoware-documentation/tree/main/docs/contributing/coding-guidelines/ros-nodes is empty.

### Possible approaches

Writing guidelines in  console-logging.md

### Definition of done

finisih writing

alanmengg, 2022-10-12:
> @alanmengg Sorry for the inconvenience. Do you have any ideas about the guidelines?

I think I can write some basic usage case using autoware examples~

kenji-miyake, 2022-10-13:
Oh, it's great! In that case, let's discuss it on your PR.

Today I found a PR that should be mentioned in the logging guideline. cc @takayuki5168 
https://github.com/autowarefoundation/autoware.universe/pull/2056

PR #283 (2022-12-26): feat: add console-logging doc
## Description

Suggestions for improvements to #216.
Resolves #212

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The Reviewers **must** check the checkboxes below before approval.

- [x] The PR follows the [pull request guidelines].

## Post-review checklist for the PR author

The PR author **must** check the checkboxes below before merging.

- [ [... cut]
closed 2022-12-26 by kenji-miyake