autowarefoundation/autoware-documentation #215: Complement of TODO list in "how-to guide"

JianKangEgon, 2022-10-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

I'm going to prepare docs about "Create an Autoware package" in "how to guide". However, I'm not sure if I understand this part correctly. Is this section to guide new user how to customize his own project based on Autoware (msgs, tools, etc..)  or instruct people how to become an Autoware code contributor?

### Purpose

Get clear of the scope.

### Possible approaches

TBD

### Definition of done

completion of docs

kenji-miyake, 2022-10-19:
@JianKangEgon Thank you for opening this.

> instruct people how to become an Autoware code contributor

I meant this by "Create an Autoware package".

> guide new user how to customize his own project based on Autoware (msgs, tools, etc..)

I feel it's like an extension of https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware-with-your-vehicle/. It's also worth writing a document.

JianKangEgon, 2022-10-20:
@kenji-miyake oh thanks for the reply, got it.

closed 2022-10-20 by JianKangEgon