autowarefoundation/autoware-documentation #266: Add guides about coordinate syatem

cyn-liu, 2022-12-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Current `coordinate-system.md`  in [directory](https://autowarefoundation.github.io/autoware-documentation/main/contributing/coding-guidelines/ros-nodes/coordinate-system/) is empty.
I plan to write a guideline about coordinate system.

### Purpose

 Write a guideline about coordinate system.

### Possible approaches

I plan to write the following:
1. The coordinate system in Autoware(for example map, base_link, sensor coordinate system).
2. An example of how to use TF2 for coordinate transformation in Autoware.

@kenji-miyake  Do you have any advice?

### Definition of done

TO DO!

kenji-miyake, 2022-12-05:
@cyn-liu I think you can refer to these documents.
https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-interfaces/components/vehicle-dimensions/
https://www.ros.org/reps/rep-0105.html

cyn-liu, 2022-12-15:
@mitsudome-r Hello:
When I write the tutorial of `coordinate system` , I encountered some problems:
what is the difference of the `camera_link` and `camera_optical_frame`?

[image]

mitsudome-r, 2022-12-20:
[image]
This image shows the difference between the camera link and camera optical.

Check: https://youtu.be/A3nw2M47K50?t=533

closed 2023-01-16 by cyn-liu