autowarefoundation/autoware-documentation #270: Add video tutorial of autoware installation

cyn-liu, 2022-12-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

I plan to add the Autoware installation video tutorial in [here](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/), but the Autoware installation tutorial document is relatively complex. Even the Docker based installation tutorial requires a lot of additional dependencies, which is very unfriendly to beginners.

Autoware Universe now mostly provides images of the `prebuilt` version [(image link)](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe). Based on this image, you do not need to download the source code of Autoware, and you can run Autoware without compiling. So can I record Autoware installation videos based on the `prebuilt` version images?

Autoware installation documentation [(here)](https://autowarefoundation.github.io/autoware-documentation/main/installation/) needs to be updated?

@kenji-miyake Do you have any suggestion?

### Purpose

Add video of Autoware installation tutorial.

### Possible approaches

Add video of Autoware installation tutorial.

### Definition of done

TO DO.

kenji-miyake, 2022-12-07:
@cyn-liu cc @mitsudome-r @xmfcx 

> So can I record Autoware installation videos based on the prebuilt version images?

Sure. FYI, there is a Japanese tutorial video by @yukkysaito. (recorded about a half year ago)
https://www.youtube.com/watch?v=iW-a7cKUxuY

I think a similar composition is good.

> Autoware installation documentation [(here)](https://autowarefoundation.github.io/autoware-documentation/main/installation/) needs to be updated?

Do you mean you want to put the tutorial video on the documentation page?
In that case, yes, I think it's a good idea.

cyn-liu, 2022-12-07:
> Do you mean you want to put the tutorial video on the documentation page? 

Yes, but I only want record video of `docker installation of autoware`.

cyn-liu, 2022-12-07:
@xmfcx Good job!
I only want record video about docker installation, but I find the document of `docker installation` is complex. If use `prebuilt image`, we don't need to install so many dependencies.

xmfcx, 2022-12-07:
> @xmfcx Good job! I only want record video about docker installation, but I find the document of `docker installation` is complex. If use `prebuilt image`, we don't need to install so many dependencies.

Thanks, sure, go ahead and create a tutorial video, we can put it to the place you've proposed.

cyn-liu, 2022-12-09:
I think the document of [Docker installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/) should be divided into two branches. One is to use the `non-prebuilt version image` installation method and the other is to use the `prebuilt version image` installation method.

The second method is very friendly for beginner to run Autoware.
@kenji-miyake **what's your opinion?**

I wrote a draft about Autoware docker installation use `prebuilt version image` .
my draft:
# Autoware installation

## Docker installation (prebuit version)

### Install docker
curl -fsSL https://get.docker.com -o get-docker.sh
sudo sh get-docker.sh
sudo groupadd docker
sudo usermod -aG docker $USER
newgrp docker

### Install nvidia container toolkit
distribution=$(. /etc/os-release;echo $ID$VERSION_ID) \
      && curl -s -L https://nvidia.github.io/libnvidia-container/gpgkey | sudo apt-key add - \
      && curl -s -L https://nvidia.github.io/libnvidia-container/$distribution/libnvidia-container.list | sudo tee /etc/apt/sources.list.d/nvidia-container-toolkit.list

sudo apt-get update
sudo apt-get install -y nvidia-docker2

sudo systemctl restart docker
test install: sudo docker run --rm --gpus all nvidia/cuda:11.0.3-base-ubuntu20.04 nvidia-smi

### Install rocker
sudo apt update && sudo apt install curl gnupg lsb-release
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg

echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(source /etc/os-release && echo $UBUNTU_CODENAME) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null

sudo apt update
sudo apt install python3-rocker

### Pull image
docker pull ghcr.io/autowarefoundation/autoware-universe:humble-latest-prebuilt (Autoware Foundation registry)

### Create the autoware_map directory for map data later.

mkdir ~/autoware_map

**NOTE**: Please put your map files in this directory.

### run containner
rocker --nvidia --x11 --privileged --user --volume $HOME/autoware_map -- ghcr.io/autowarefoundation/autoware-universe:humble-latest-prebuilt

### run autoware
e.g., running planning simulator:
ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit

**NOTE**: The Autoware code located in the `/autoware` directory in the container , if you want to modify the code, please enter this directory.

### Build the workspace
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

cyn-liu, 2022-12-12:
@kenji-miyake 
Do you mean to put this installation tutorial on the [introduction page](https://autowarefoundation.github.io/autoware-documentation/main/) as a quick start?

kenji-miyake, 2022-12-12:
@cyn-liu Sorry for my unclear sentence. I meant just putting the link to a quick start there.
I think it's good to put the content on https://autowarefoundation.github.io/autoware-documentation/main/tutorials/quickstart (a new page).
What do you think? cc @xmfcx

cyn-liu, 2022-12-13:
@xmfcx

xmfcx, 2022-12-13:
> I think the document of [Docker installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/) should be divided into two branches. One is [...]

https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/ 

Here I think installing the dependencies is common for both methods. I think you should improve existing dependency installation methods instead of rewriting them.

And I agree that you can split the `prebuilt` and `non-prebuilt`. Please make 2 pages under Docker installation for them.

So user goes:
Docker installation -> *sets up environment, installs dependencies* -> There are 2 options to continue: `prebuilt` and `non-prebuilt`.

Is this ok with you @cyn-liu ?

kenji-miyake, 2022-12-13:
@xmfcx Thank you for your comments. The reason why I said we should split pages is that the target readers of the pages are different. `quickstart` is for beginners or runtime-only users and the existing `docker-installation` is for developers.
It's okay for me to write both information on the `docker-installation` page, but I believe it's better to have another `quickstart` page for beginners.
Is that not good?

xmfcx, 2022-12-13:
@kenji-miyake I think developers can use prebuilt version to edit & compile autoware too,
as @cyn-liu shared in the post:
> ### Build the workspace
> colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

And our docker documentation also have:
> 5. Build the workspace.
> colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

It would be confusing to have these 2 pages in different sections since they both achieve the same result.

kenji-miyake, 2022-12-13:
@xmfcx The prebuilt version will be simplified in the future by https://github.com/autowarefoundation/autoware/pull/2922. After that, developers cannot use it for development because the image has only runtime dependencies. I imagined that.

xmfcx, 2022-12-13:
@kenji-miyake I will comment there to separate `prebuilt` and `release` containers.

doganulus, 2022-12-13:
I think Autoware documentation should deliberately avoid any suggestion or recommendation to install Autoware on the local system. Let's put our effort to prepare an Autoware `devel` image (aka devcontainer).

Let the user pull the development image and mount Autoware source folder. The current images comes with source as far as remember. Encourage the developer as much as possible to work inside the container.

xmfcx, 2022-12-13:
> I think Autoware documentation should deliberately avoid any suggestion or recommendation to install Autoware on the local system. Let's put our effort to prepare an Autoware `devel` image (aka devcontainer).
> 
> Let the user pull the development image and mount Autoware source folder. The current image comes with source as far as remember. Encourage the developer as much as possible to work inside the container.

@doganulus There are many core developers of the project who uses native installation method exclusively due to performance and simplicity purposes. Not everyone wants the overhead and added complexity of dealing with containers.

We have a docker image and instructions to use it, if you have an issue with how it works, how it is constructed or how it can be improved, please create a separate issue for it.

doganulus, 2022-12-13:
An experienced user can always install Autoware on their local system. I do not think we cannot prevent that. But this issue is about everyone else, especially the newcomer. Therefore let's focus on the people who want to start developing Autoware.

Still developing inside the container is very valuable for experienced core developers, I believe.

cyn-liu, 2022-12-14:
> And I agree that you can split the `prebuilt` and `non-prebuilt`. Please make 2 pages under Docker installation for them.
> 
> So user goes: Docker installation -> _sets up environment, installs dependencies_ -> There are 2 options to continue: `prebuilt` and `non-prebuilt`.

I agree that.
I think it's better to keep them together as they are 2 branches of a common source.

cyn-liu, 2022-12-15:
> The prebuilt version will be simplified in the future by [autowarefoundation/autoware#2922](https://github.com/autowarefoundation/autoware/pull/2922). After that, developers cannot use it for development because the image has only runtime dependencies. I imagined that.

@xmfcx  @kenji-miyake 
I have confirmed this problem. The image of the future `prebuild version` can only run Autoware, and cannot use it for development.

I plan to divide the `docker installation` into two files. One is the `docker installation with prebuild image`, and the other is the `docker installation with devel image`, and explain the usage scenarios of the two methods.

xmfcx, 2022-12-16:
@cyn-liu I've created this training video series: https://github.com/orgs/autowarefoundation/discussions/3141

I didn't cover docker parts, I hope you will cover them here (`･ω･´)

closed 2023-02-24 by cyn-liu