autowarefoundation/autoware-documentation #293: Add documentation for task scheduling

xmfcx, 2023-01-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Following [these comments](https://github.com/autowarefoundation/autoware-documentation/pull/282#issuecomment-1362562226) in the https://github.com/autowarefoundation/autoware-documentation/pull/282
> We TIER Ⅳ members are now working on scheduling issues on Linux platforms. Our opinion is that the task scheduling document should be about how to configure the settings of a kernel scheduler to run Autoware in a predictable way. Based on workload analysis and detailed analysis of the implementation of the Linux kernel, we plan to publish a specialized functionality for the task scheduling of Autoware on Linux and the documentation about it.

> Additionally, we need to deal with double scheduling problems (ROS2 executer and kernel scheduler) and scheduler settings for specialized operating systems. We are also struggling with these problems. User-defined and centralized task scheduler will be developed specialized for Autoware.

> As a result, we’d like to keep this document blank until the functionality as explained above become fixed.

### Purpose

Fill in the documentation for task scheduling.

### Possible approaches

Fill in the documentation for task scheduling.

### Definition of done

Fill in the documentation for task scheduling.

xmfcx, 2023-01-06:
https://github.com/autowarefoundation/autoware-documentation/pull/282 can be used as a guide when this issue is being worked on.

kenji-miyake, 2023-01-07:
@xmfcx Thank you for creating this issue!
I think how we design callback functions should be documented in the page.
For example as follows:
- When should we use multiple callback groups?
- What kind of callbacks are there?
  - Just subscribing data and buffering it.
  - Subscribing data and applying some preprocessing.
  - Trigger the main process.
  - etc.