autowarefoundation/autoware-documentation #296: Add naming convention for messages

xmfcx, 2023-01-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Follow up from https://github.com/autowarefoundation/autoware_msgs/pull/48#discussion_r1068235149

cc. @isamu-takagi 

### Purpose

To make it clear how we name plural messages.

from @isamu-takagi 
> Many of the ros messages use Array sufix, so I think it's better to match them.
>
> ```
> diagnostic_msgs/msg/DiagnosticArray
> geometry_msgs/msg/PoseArray
> visualization_msgs/msg/MarkerArray
> ```

### Possible approaches

Edit https://autowarefoundation.github.io/autoware-documentation/main/contributing/coding-guidelines/ros-nodes/message-guidelines/

### Definition of done

[Docs ](https://autowarefoundation.github.io/autoware-documentation/main/contributing/coding-guidelines/ros-nodes/message-guidelines/) is updated.

isamu-takagi, 2023-01-16:
I created a PR https://github.com/autowarefoundation/autoware-documentation/pull/297.
[Since the message name has not been discussed](https://github.com/autowarefoundation/autoware/discussions/2824#discussioncomment-3513433), other than suffix is under construction.

xmfcx, 2023-01-16:
This issue was only for plural types, so I'm closing this. Thanks @isamu-takagi !

closed 2023-01-16 by xmfcx