autowarefoundation/autoware-documentation #302: Image and Latex Equations are not properly loaded/rendered

ahmeddesokyebrahim, 2023-01-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

In the [Behavior path planner path generation documentation ](https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_path_planner/behavior_path_planner_path_generation/), there are some images and latex equations are not loaded properly. 

However, the same image and equations are shown correctly for the same document but in [github readme](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/behavior_path_planner_path_generation.md)

I will be very beneficial if this can be fixed in autoware documentation as well. 

### Purpose

Fix the image and latex equations loading for having better reading and understanding of autware documentation

### Possible approaches

- Check if autoware.github.io can access the images to be loaded (could be a broken path) 
- Check if autoware.github.io have proper rendering plugins for latex equations

### Definition of done

To see the same output properly shown in [Behavior path planner path generation documentation ](https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_path_planner/behavior_path_planner_path_generation/), as in [github readme](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/behavior_path_planner_path_generation.md)

mehmetdogru, 2023-01-18:
cc: @kenji-miyake

kenji-miyake, 2023-01-18:
Related: https://github.com/autowarefoundation/autoware.universe/issues/1568

This is because of the difference between the Markdown grammar of "GitHub" and "Material for MkDocs".

For images, the format `<img src="./image/path_shifter.png" width="800">` cannot be used.
Please use `[path_shifter](./image/path_shifter.png)` instead.

For Latex, we need to use `$$` instead of ` ```math `.

cc @TakaHoribe (The author of https://github.com/autowarefoundation/autoware.universe/pull/2428)

mehmetdogru, 2023-01-26:
@ahmeddesokyebrahim could you please take a look at the proposed solution.

ahmeddesokyebrahim, 2023-02-03:
I have tested the proposed solution by @kenji-miyake in this branch :
https://github.com/ahmeddesokyebrahim/autoware.universe/blob/ahmedebrahim/aw-doc/path-generation-readme/planning/behavior_path_planner/behavior_path_planner_path_generation.md

And it looks like that we either sacrifice github doc or github readme.

I will not create a PR and close the issue with this comment.

closed 2023-02-03 by ahmeddesokyebrahim