autowarefoundation/autoware-documentation #351: DevOps Dojo: ROS Node Configuration - JSON Schema to MD-Table for Web Documentation

kasperornmeck, 2023-04-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Automatically generate MD table in web documentation from `*.schema.json* file for ROS node parameters.

### Purpose

Using the JSON Schema to generate the parameter table for the web documentation removes the need to duplicate effort.

### Possible approaches

```json
{
  "$schema": "http://json-schema.org/draft-07/schema#",
  "title": "Parameters for Lidar Apollo Segmentation TVM Nodes",
  "type": "object",|
  "definitions": {
    "lidar_apollo_segmentation_tvm_nodes": {
      "type": "object",
      "properties": {
        "range": {
          "type": "number",
          "default": 90,
          "exclusiveMinimum": 0,
          "description": "The range of the 2D grid with respect to the origin."
        }
      },
      "required": [
        "range"
      ]
    }
  },
  "properties": {
    "/**": {
      "type": "object",
      "properties": {
        "ros__parameters": {
          "$ref": "#/definitions/lidar_apollo_segmentation_tvm_nodes"
        }
      },
      "required": ["ros__parameters"]
    }
  },
  "required": ["/**"]
}
```
should be generated to
| Parameter Name | Type | Default | Description |
| --- | --- | --- | --- |
| range | number | 90 | The range of the 2D grid with respect to the origin. |

### Definition of done

Once the JSON Schema is generated into a table in the web documentation, see example above.

kaspermeck-arm, 2023-08-17:
@mitsudome-r is working on this, see:
- https://github.com/autowarefoundation/autoware.universe/pull/4656/files

kaspermeck-arm, 2023-09-07:
Add the following line in the `README.md` file:
* https://github.com/autowarefoundation/autoware.universe/blob/main/perception/lidar_apollo_segmentation_tvm_nodes/README.md#parameters

which generates the following table:
* https://autowarefoundation.github.io/autoware.universe/main/perception/lidar_apollo_segmentation_tvm_nodes/#parameters

Pending:
* @mitsudome-r - update the parameter documentation to explain how the table generator works

kaspermeck-arm, 2023-11-06:
Completed by Mitsudome-san a while back.

closed 2023-11-06 by kaspermeck-arm

VRichardJP, 2024-04-04:
Sorry for the move, GitHub does not like touch screens