autowarefoundation/autoware-documentation #396: Add documentation for Creating vehicle and sensor description

Naophis, 2023-06-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

[Creating vehilce and sensor description](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-description/creating-vehicle-and-sensor-description/) is empty,  I'm going to add document it.

### Purpose

Adding specific instructions for creating vehicle and sensor descriptions.

### Possible approaches

The following three explanations.
* Description of vehicle_description configuration.
* Description of sensor_description configuration.
* Description of individual_parameter, how to be used in sensor_description.

### Definition of done

write document and merge it.

Naophis, 2023-06-16:
@kenji-miyake @mitsudome-r 
May I add to the document with this direction?

kenji-miyake, 2023-06-16:
@Naophis Of course, thank you!

commit fe80bc38 (2023-06-16): docs(how-to-guides): Creating vehicle and sensor description.
Fixes #396
commit 31f01dc9 (2023-06-16): docs(how-to-guides): Creating vehicle and sensor description.
Fixes #396
commit 039b9f6e (2023-06-16): docs(how-to-guides): Creating vehicle and sensor description.
Fixes #396
commit 190caa02 (2023-06-18): docs(how-to-guides): add creating vehicle and sensor description. (#399)

* docs(how-to-guides): Creating vehicle and sensor description.
Fixes #396

* style(pre-commit): autofix

* fix: typo

* fix: typo

---------
closed 2023-06-18 by Naophis