autowarefoundation/autoware-documentation #548: Add docs for guiding the developers on finding, diagnosing, and fixing scenarios in Autoware

xmfcx, 2024-05-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

This issue aims to provide comprehensive guidance for developers interested in contributing to Autoware by identifying, troubleshooting, and resolving issues within various test scenarios.

It will serve as a resource for planning developers looking for jobs, new-comers evaluating Autoware's test success, and contributors aiming to fix broken scenarios without causing regressions.

### Purpose

The purpose of this documentation is to streamline the process for developers to engage with Autoware's testing framework effectively.

It will enable contributors to easily find scenarios, understand their results, reproduce issues, and confidently submit fixes ensuring no disruption to other parts of the software.

### Possible approaches

- **Scenario Discovery:** Document the locations and resources where developers can find scenarios.
- **Success Evaluation:** Outline the criteria and steps to determine if a scenario has succeeded or failed.
- **Failure Analysis:** Provide guidelines on how to analyze the reasons behind scenario failures.
- **Environment Setup:** Include detailed instructions on setting up the correct version of Autoware for reproducing scenarios.
- **Regression Testing:** Explain how to test changes locally to ensure no other scenarios are affected before pushing updates.

### Definition of done

- [ ] Comprehensive guidelines on where and how to find scenarios are documented.
- [ ] Criteria for evaluating the success of a scenario are clearly defined.
- [ ] Instructions on identifying and analyzing the reasons for scenario failures are provided.
- [ ] Steps to reproduce the exact scenario environment are detailed.
- [ ] Procedures for running regression tests to ensure no disruptions are described.

xmfcx, 2024-05-13:
I've updated the Possible approaches section.

brkay54, 2024-08-24:
Documentation added, I forgot to close it.

brkay54, 2024-08-24:
The PR has been merged, so I am closing this issue.

closed 2024-08-24 by brkay54